cvg/nice-slam #2: Matrix Inverse Error

kyleleey, 2022-04-19:
Great work. 

When I try this code, I met a same error for several senarios:
```bash
Process Process-1:   
Traceback (most recent call last):                                                                                                                                                                                  
   File "/home/zzl/miniconda3/envs/nice-slam/lib/python3.7/multiprocessing/process.py", line 297, in _bootstrap                                                                                               
      self.run()                                                                                                                                                                                                      
   File "/home/zzl/miniconda3/envs/nice-slam/lib/python3.7/multiprocessing/process.py", line 99, in run                                                                                                        
      self._target(*self._args, **self._kwargs)                                                                                                                                                                      
   File "/home/zzl/workspace/nice-slam/src/NICE_SLAM.py", line 263, in tracking                                                                                                                                
      self.tracker.run()                                                                                                                                                                                              
   File "/home/zzl/workspace/nice-slam/src/Tracker.py", line 195, in run                                                                                                                                       
     device).float().inverse()         
torch._C._LinAlgError: cusolver error: CUSOLVER_STATUS_EXECUTION_FAILED, when calling `cusolverDnSgetrf( handle, m, n, dA, ldda, static_cast<float*>(dataPtr.get()), ipiv, info)`. This error may appear if the input matrix contains NaN.
```

So far, this happens for imap in Apartment and nice-slam in TUM_RGBD/freiburg1 (nice-slam in Apartment and imap in TUM_RGBD/freiburg1 are ok). I haven't try other scenarios yet.

This error usually happens in first five frames optimization and is reproducible. I haven't change any code yet, so I wonder why the matrix that can't be inversed emerges.

Zzh2000, 2022-04-20:
Hi Zizhang,

Thanks for your interest in our work!
I just tested the code on a brand new machine following the step-by-step description in README.md, and it does not have the error you said. Could you please give more information, for example, your environment? Also, it would be nice of you to print out the matrix it cannot inverse.

Best,
Zihan

kyleleey, 2022-04-20:
Hi Zihan,

Thanks for reply. This error occurs for this line of [code](https://github.com/cvg/nice-slam/blob/master/src/Tracker.py#L194):
```python
delta = pre_c2w@self.estimate_c2w_list[idx-2].to(device).float().inverse()
```

I try to print the matrix that can't be inversed, but I find it is a matrix that **can** be inversed, for example for imap* in CoFusion/room4, at Frame 2, the matrix to be inversed is:
```bash
tensor([[ 9.9996e-01,  6.4982e-03,  5.0393e-03,  3.8073e-03],
        [-6.5110e-03,  9.9997e-01,  2.5397e-03,  1.5920e-03],
        [-5.0227e-03, -2.5724e-03,  9.9999e-01,  3.3164e-04],
        [ 0.0000e+00,  0.0000e+00,  0.0000e+00,  1.0000e+00]])
```

I ran all the experiments that failed before on different GPU settings, and find that when using RTX3090 with CUDA 11.2 the computation is fine, but when on GTX1080 with CUDA 11.0 this error occurs. So far I think the CUDA version is the reason.

Zzh2000, 2022-04-20:
I see, maybe GTX1080 is a bit old and has some compatibility issues with CUDA 11.0 and some PyTorch functions. Glad to hear you are able to run it now on a new machine.

closed 2022-04-20 by Zzh2000

kyleleey, 2022-04-22:
Hi

The matrix inverse computation also fails for RTX TITAN with CUDA 11.1, but if move the matrix to cpu will not cause any error. So I change the L195 to following and it seems to be an appropriate solution with no performance decreasing.

```python
delta = pre_c2w@self.estimate_c2w_list[idx-2].float().inverse().to(device)
```