cvg/nice-slam #3: IndexError: list index out of range

BIT-DYN, 2022-04-20:
Hi, thank you  for such an excellent work.

When I try to run the demo, I get an error like this:
```
Process Process-3:
Traceback (most recent call last):
  File "/home/chenguangyan/anaconda3/envs/nice-slam/lib/python3.7/multiprocessing/process.py", line 297, in _bootstrap
    self.run()
  File "/home/chenguangyan/anaconda3/envs/nice-slam/lib/python3.7/multiprocessing/process.py", line 99, in run
    self._target(*self._args, **self._kwargs)
  File "/home/chenguangyan/dengyinan/Nice-SLAM/src/NICE_SLAM.py", line 285, in coarse_mapping
    self.coarse_mapper.run()
  File "/home/chenguangyan/dengyinan/Nice-SLAM/src/Mapper.py", line 544, in run
    idx, gt_color, gt_depth, gt_c2w = self.frame_reader[0]
  File "/home/chenguangyan/dengyinan/Nice-SLAM/src/utils/datasets.py", line 78, in __getitem__
    color_path = self.color_paths[index]
IndexError: list index out of range
```

I don't think I changed any code. If you can have some suggestions, I would be very grateful.

BIT-DYN, 2022-04-20:
The problem has been resolved.

closed 2022-04-20 by BIT-DYN

LongruiDong, 2022-05-18:
Hi ,

I also met this error when run scannet data while other data succeed to run...

how do you solve it?

Zzh2000, 2022-05-18:
Hi, you can double-check if you have downloaded the dataset to the correct directory, we did not provide the scannet data (for copyright reasons) so you have to download it on your own, and put it in the right place.