cvg/nice-slam #4: If we don't know the bounding of the scene, can this method works?

endlesswho, 2022-04-24:
The code shows us that we need to use open3d reconstruction pipline for scene reconstruction. However, if we don't know the bounding of the scene, can this method works or any ideas to estimate the size of the scene?

Zzh2000, 2022-04-24:
Hi, you can give an Specify the bound of the scene. If no ground truth camera pose is given, we construct world coordinates on the first frame. The X-axis is from left to right, Y-axis is from down to up, Z-axis is from front to back.

Zzh2000, 2022-04-24:
Hi, you can give an estimate of the scene bound without the need to run open3d, please see item 3 in the README.md about running with your own sequence: 

> 3. Specify the bound of the scene. If no ground truth camera pose is given, we construct world coordinates on the first frame. The X-axis is from left to right, Y-axis is from down to up, Z-axis is from front to back.

closed 2022-04-24 by Zzh2000