cvg/nice-slam #7: Evaluates the occupancy and/or color value for the points.

mabaorui, 2022-04-27:
https://github.com/cvg/nice-slam/blob/645b53af3dc95b4b348de70e759943f7228a61ca/src/utils/Renderer.py#L57
https://github.com/cvg/nice-slam/blob/645b53af3dc95b4b348de70e759943f7228a61ca/src/common.py#L239
Hi，thanks for such a great work, I was wondering why the value of points out of bound  are set to 100,  if these points are set to 100, isn't the weight at this point very large during volume rendering?

Zzh2000, 2022-04-27:
Hi,
thanks for your interest in our work. Actually, you can check the rendering equation in NeRF, it considers the occlusion in the front. You can print out the weights, they will not be large.

closed 2022-04-27 by Zzh2000

mabaorui, 2022-04-28:
Hi，
Thanks for your reply, I understand how nerf works, the situation you said requires the points out of bound  to be farther along this ray during rendering, in that case, why not set it to -100 or don't deal with it, it won't make a big difference to the final weight for rendering.

Zzh2000, 2022-05-01:
Hi, 
Yes. I think +-100 or don't deal with it will not affect the rendering.

mabaorui, 2022-05-01:
I get it, thanks!