cvg/nice-slam #8: Is there a need for large CUDA memory?

jarvishou829, 2022-05-04:
Hi, I tried to run the nice-slam on TUM_GRBD freiburg1_desk sequence according to the readme file, but got CUDA error: out of memory. 
`RuntimeError: CUDA out of memory. Tried to allocate 86.00 MiB (GPU 0; 5.80 GiB total capacity; 686.18 MiB already allocated; 98.94 MiB free; 780.00 MiB reserved in total by PyTorch) If reserved memory is >> allocated memory try setting max_split_size_mb to avoid fragmentation.  See documentation for Memory Management and PYTORCH_CUDA_ALLOC_CONF`
My device is 3060 laptop 6G
Driver Version: 470.63.01
CUDA Version: 11.4

Zzh2000, 2022-05-04:
Hi, our scene representation is neural implicit based (like NeRF) which requires desktop/server GPU like 3090, 2080ti, TITAN X to run it smoothly.

closed 2022-05-04 by Zzh2000

Jing-lun, 2022-06-10:
> Hi, our scene representation is neural implicit based (like NeRF) which requires desktop/server GPU like 3090, 2080ti, TITAN X to run it smoothly.

Hi @Zzh2000 Zihan, I encountered a strange problem that when training on the apartment dataset, I always have a Cuda out of memory error. However, my GPU is also a 3090 like that you used in your paper and when I carefully watch the nvidia-smi during the training, I never found a moment that the GPU memory was all occupied, especially considering 3090 has 24 GB RAM.
This error kept popping up at 105 frame's mapping on the apartment dataset training, so if you could provide me more insights on how to train your apartment dataset, such as how did you set up your hyper parameters, I would appreciate it a lot!

Zzh2000, 2022-06-11:
Hi, that is very strange, the apartment dataset is also runnable on 2080ti, TITAN X. I did not encounter OOM problems before. The hyper-parameters are the same as in the config file. Could you try to run it on another computer to check?

Jing-lun, 2022-06-12:
> Hi, that is very strange, the apartment dataset is also runnable on 2080ti, TITAN X. I did not encounter OOM problems before. The hyper-parameters are the same as in the config file. Could you try to run it on another computer to check?

It is a strange bug because I never watched a single moment of running the apartment dataset requiring more than 12 GB RAM in GPU. However, after I ran this dataset on a 2080ti, I met the thread_monitor resource issue... it showed that
> thread_monitor resource temporarily unavailable in pthread_create

Zzh2000, 2022-06-19:
Hi, I think it might be because of the system configuration, instead of the Conda environment. Which seems to be hard to debug. Several master students in ETH worked on extensions of NICE-SLAM and they successfully configured on Amazon AWS and AutoDL. AutoDL supports mirror sharing. Let me know if you need that :)

Lilyo, 2022-07-26:
Hi @Jing-lun , 

I got the same issue, and I have solved by set `num_workers=0` in Dataloader of `src/Tracker.py`.

garriton, 2022-10-27:
> num_workers

I got the same issue, and this set solves, thanks

garriton, 2022-10-27:
> Hi @Jing-lun ,
> 
> I got the same issue, and I have solved by setting `num_workers=0` in Dataloader of `src/Tracker.py`.

I got the same issue, and this set solves, thanks

Yiiii19, 2023-10-02:
> Hi @Jing-lun ,
> 
> I got the same issue, and I have solved by setting `num_workers=0` in Dataloader of `src/Tracker.py`.

Thanks for pointing this. Before applying your change I can only run until 105 frame, after using your change i can achieve 215 frame. I just want to try the appartment run.py file. Do you have any idea about that? I used 3090 which has 24 GB.
Error message:
RuntimeError: CUDA out of memory. Tried to allocate 30.00 MiB (GPU 0; 23.68 GiB total capacity; 1.03 GiB already allocated; 15.07 GiB free; 1.08 GiB reserved in total by PyTorch)

Zoey6677, 2024-02-26:
> It is a strange bug because I never watched a single moment of running the apartment dataset requiring more than 12 GB RAM in GPU. However, after I ran this dataset on a 2080ti, I met the [...]

@Jing-lun Hello, have you gotten the AutoDL mirror that the author said? I created an instance myself in AutoDL but can't visualize it with open3d warning, which seems to be hard to fix. So I think the AutoDL mirror sharing provided by the author will help me a lot. Thank you!