cvg/nice-slam #10: about camera pose

anooooob, 2022-05-17:
hi, can you explain why the camera pose[:3, 1:2] should be multiply by -1?
e.g. in src/utils/datasets.py 
            c2w[:3, 1] *= -1.0
            c2w[:3, 2] *= -1.0

Zzh2000, 2022-05-17:
Hi，
we use the camera coordinate similar to NeRF-PyTorch's code, that is X-axis is from left to right, Y-axis is from down to up, Z-axis is from front to back. However, it seems that all the datasets we ran on are using Y-axis from up to down, Z-axis from back to front, so *-1 is needed. However, please note that other datasets may use other camera axis conventions.

closed 2022-05-17 by Zzh2000

ckLibra, 2022-10-30:
> Hi， we use the camera coordinate similar to NeRF-PyTorch's code, that is X-axis is from left to right, Y-axis is from down to up, Z-axis is from front to back. However, it seems that all the [...]

If so, why did you just convert the rotation but not also the translation?

VladimirYugay, 2023-05-08:
@ckLibra did you figure it out?