cvg/nice-slam #11: hoiw to set scene bound when run outdoor scene?

LongruiDong, 2022-05-24:
Hi~

Thans for sharing this amazing work！

I am testing on outdoor scene : vkitti2，but I haven't figured out how to set bound in mapping.

For example, according ground truth poses at word frame(first frame), I can get the boundary of whole trajectory, e.g. [-137,2],[-1,10],[0,214],

then how to properly decide the bound and marching_cubes_bound parameters in order to get reasonable mesh result?

Zzh2000, 2022-05-24:
hi, thanks for your interest in our work.
The bound/marching_cubes_bound should be set as the bound of the scene you want to reconstruct (the bound doesn't need to be accurate at all, can be very coarse), the bound of the trajectory may not be enough. We focus on indoor scenes, so there might be some issues if you directly run it on outdoor scenes. Good luck!

closed 2022-05-24 by Zzh2000

LongruiDong, 2022-05-30:
> hi, thanks for your interest in our work. The bound/marching_cubes_bound should be set as the bound of the scene you want to reconstruct (the bound doesn't need to be accurate at all, can be very [...]

Thanks for your reply.

Do you think it is necessary to retrain those 3 MLP decoders if run on outdoor data?

Geniussh, 2023-02-09:
@LongruiDong Did you retrain the decoder later? How is the performance on outdoor data?