cvg/nice-slam #12: Qt issue when test Apartment data

LongruiDong, 2022-05-24:
Hi:

I have successfully test **all other provided data** in this repo.
However, only when test Apartment, met following error:

```
Coarse Mapping Frame  50

This application failed to start because it could not find or load the Qt platform plugin "xcb"
in "/home/dlr/anaconda3/envs/nice-slam/lib/python3.7/site-packages/cv2/qt/plugins".

Available platform plugins are: eglfs, minimal, minimalegl, offscreen, vnc, xcb.

Reinstalling the application may fix this problem.
QStandardPaths: wrong ownership on runtime directory /usr/lib/, 0 instead of 1000
Process Process-3:
Traceback (most recent call last):
  File "/home/dlr/anaconda3/envs/nice-slam/lib/python3.7/multiprocessing/process.py", line 297, in _bootstrap
    self.run()
  File "/home/dlr/anaconda3/envs/nice-slam/lib/python3.7/multiprocessing/process.py", line 99, in run
    self._target(*self._args, **self._kwargs)
  File "/home/dlr/Project/nice-slam/src/NICE_SLAM.py", line 283, in coarse_mapping
    self.coarse_mapper.run()
  File "/home/dlr/Project/nice-slam/src/Mapper.py", line 606, in run
    gt_c2w, self.keyframe_dict, self.keyframe_list, cur_c2w=cur_c2w)
  File "/home/dlr/Project/nice-slam/src/Mapper.py", line 428, in optimize_map
    idx, joint_iter, cur_gt_depth, cur_gt_color, cur_c2w, self.c, self.decoders)
  File "/home/dlr/Project/nice-slam/src/utils/Visualizer.py", line 85, in vis
    gt_color_np = np.clip(gt_color_np, 0, 1)
  File "<__array_function__ internals>", line 6, in clip
  File "/home/dlr/anaconda3/envs/nice-slam/lib/python3.7/site-packages/numpy/core/fromnumeric.py", line 2115, in clip
    return _wrapfunc(a, 'clip', a_min, a_max, out=out, **kwargs)
  File "/home/dlr/anaconda3/envs/nice-slam/lib/python3.7/site-packages/numpy/core/fromnumeric.py", line 57, in _wrapfunc
    return bound(*args, **kwds)
  File "/home/dlr/anaconda3/envs/nice-slam/lib/python3.7/site-packages/numpy/core/_methods.py", line 160, in _clip
    um.clip, a, min, max, out=out, casting=casting, **kwargs)
  File "/home/dlr/anaconda3/envs/nice-slam/lib/python3.7/site-packages/numpy/core/_methods.py", line 113, in _clip_dep_invoke_with_casting
    return ufunc(*args, out=out, **kwargs)
numpy.core._exceptions._ArrayMemoryError: Unable to allocate 21.1 MiB for an array with shape (720, 1280, 3) and data type float64

```

then the process just stops...

I have not changed the yaml in config/.. so strange here

WeiCaow, 2022-05-30:
Hi, I met this problem when I run the script for scannet, do you maybe have any idea.....?
[image]

LongruiDong, 2022-05-30:
> Hi, I met this problem when I run the script for scannet, do you maybe have any idea.....? [image]

https://github.com/cvg/nice-slam/issues/3#issuecomment-1130051621

maybe it is the same situation as mine.

make sure your structure of scannet is as follows:

[image]

closed 2022-06-03 by pengsongyou

Jiazxzx, 2022-08-12:
Same question. 
And the number is the same to yours:

> numpy.core._exceptions._ArrayMemoryError: Unable to allocate 21.1 MiB for an array with shape (720, 1280, 3) and data type float64

Have you solved it?

arqam-ai, 2022-10-19:
Hi, i have solved it following this method from stackoverflow
https://stackoverflow.com/questions/57507832/unable-to-allocate-array-with-shape-and-data-type

shaoxiang777, 2023-05-19:
> Same problem. And the number is the same to yours:
> 
> > numpy.core._exceptions._ArrayMemoryError: Unable to allocate 21.1 MiB for an array with shape (720, 1280, 3) and data type float64
> 
> Have you solved it?

Same problem. How do you solve it?

I collect my own dataset using Azure Kinect depth camera. It works well. 

Actually the dataset Apartment was sampled also by same depth camera. It was really wired.  @Jiazxzx @LongruiDong

limshoonkit, 2023-10-11:
Encounter similar issue.

The opencv-python conda comes with its own pyqt5.
What i did was removing all conda installed qt, opencv-python and matplotlib
conda remove pyqt matplotlib opencv-python
pip uninstall pyqt5 matplotlib opencv-python

then proceed with
pip install matplotlib opencv-python