cvg/nice-slam #13: Download url may not working

Jing-lun, 2022-06-06:
Hi @Zzh2000 @pengsongyou, cheers to your great work!

However, when I tried to download some of the demos/inferences, some of the download scripts returned
> HTTP request sent, awaiting response... 403 Forbidden
And it's hard for foreign users to use hecaiyun. 

Would you mind update the url for the files such as: cull_replica_mesh.zip, Demo.zip, etc.

Thanks

pengsongyou, 2022-06-07:
Hi @Jing-lun sorry for the inconvenience. The server for keeping data is currently down. We are investigating the reason. Will keep you posted.

pengsongyou, 2022-06-07:
Hi @Jing-lun the issue should be fixed already, please try again.

closed 2022-06-07 by pengsongyou

Jing-lun, 2022-06-07:
> Hi @Jing-lun the issue should be fixed already, please try again.

Great, thanks for your prompt reply!