cvg/nice-slam #15: inversion could not be completed because the imput matrix is singular

zzw-ziwei, 2022-06-24:
in Tracker.py, at line 195, prc_c2w divide the estimated_c2w, however estimated_c2w has 0 norm,  thus uninvereble, anyone knows how to fix it?

I test nice-slam on a 3080TI LAPTOP GPU, and on Apartment dataset and Replica room0 dataset, the problem remains.

If I set "const_speed_assumption" to "False", there's no need to do inversion, and the above problem disappeared, but the Depth residual and RGB residual output just as the same as the input image as below.

[image]
 so, Why? And how to fix it?

zzw-ziwei, 2022-06-24:
I test nice-slam on windows11 with cuda11.3, the singular matrix is actually a 4X4 zero matrix, this value should not be divided. With debugging, I remove "=" at line 193, thus the first estimate_c2w is replaced by invertable second one, and the program goes on. But I don't know if I have it right.

tfy14esa, 2022-06-24:
Hi zzw-ziwei,
I have tested NICE-SLAM on a linux device without issues. I have also not seen the behavior you are seeing for the generated depth and RGB. 

Sorry that I can not be of help, but just thought I'd write that it might be system specific that you get this error?

Zzh2000, 2022-06-27:
I also think it might be system specific. Let me know if you need the mirror in https://github.com/cvg/nice-slam/issues/8#issuecomment-1159638503. I can try to ask them to share the mirror with you.

closed 2022-06-27 by Zzh2000