cvg/nice-slam #19: About projection.

calmelo, 2022-06-29:
In Mapper.py,
in get_mask_from_c2w:
cam_cord[:, 0] *= -1
why are x of cam_cord needed to multiply by -1?
Besides, 
mask = mask & (0 <= -z[:, :, 0]) & (-z[:, :, 0] <= depths+0.5)
why do you use the -z[:, :, 0] rather than z[:, :, 0]?

in keyframe_selection_overlap:
mask = mask & (z[:, :, 0] < 0)
why (z[:, :, 0] < 0)?

Zzh2000, 2022-06-29:
We follow the nerf-pytorch code convention and define the z-axis as "front to back".

closed 2022-06-29 by Zzh2000

calmelo, 2022-06-30:
> We follow the nerf-pytorch code convention and define the z-axis as "front to back".

Why not cam_cord[:, 2] *= -1 but cam_cord[:, 0] *= -1?

calmelo, 2022-06-30:
> We follow the nerf-pytorch code convention and define the z-axis as "front to back".

cam_cord[:, 0] *= -1是为了让计算得到的相机坐标的x, y, z在数据集的坐标系中均取反值吗？为什么不把cam_cord[:, 1] *= -1, cam_cord[:, 2] *= -1, 然后将计算得到的相机坐标转到数据集的坐标系中呢？
取反值之后, 因为u=x/z, v=y/z, 所以得到的uv最后就在数据集图片的坐标系内，然后z取反转到数据集图片坐标系中。
不知道我的理解正不正确，恳请大佬赐教！

h8c2, 2022-06-30:
> 

我的理解是在代码相机坐标系下表达的坐标都是右上后的（正常相机坐标系为右下前，这里在load_pose就取反了），但是内参矩阵是右下前的，要变到像素坐标系就将第一列乘以负一，得到正常的像素坐标系取值范围