cvg/nice-slam #22: How did you get the bound of each scene of Scannet here?

WeiCaow, 2022-07-20:
[image]

Zzh2000, 2022-07-22:
Hi, 
you can get it by finding the xmin, xmax, ymin, ymax, zmin, zmax of vertices of the gt mesh. 
If you want to run it on your own dataset, you can set the bound using the method discussed in #4 .

closed 2022-07-22 by Zzh2000

yashbhalgat, 2022-12-29:
Hi @Zzh2000, when I use the gt mesh for `scene0000_00`, i.e. `scene0000_00_vh_clean.ply`, I get the bounds as:
```
min bounds: [-0.01763421, -0.00179759, -0.00025371]
max bounds: [10.374244,  8.737977,  3.029314]
```

In the config file [here](https://github.com/cvg/nice-slam/blob/7af15cc33729aa5a8ca052908d96f495e34ab34c/configs/ScanNet/scene0000.yaml), you use `bound: [[-2.0,11.0],[-2.0,11.5],[-2.0,5.5]]`.

I am wondering why there is a difference between these two. It would be helpful to understand how you calculated those values. 

Thanks,
Yash

Zzh2000, 2022-12-29:
Hi,
here I just use a very rough estimate, e.g. imagine myself in the scene origin and roughly estimate the span along x,y,z axis. I don't want to be accused of using gt information, so I did not provide the system with a very accurate bound. 
Best,
Zihan

yashbhalgat, 2022-12-29:
Hi Zihan, that makes sense. Thank you for the prompt response!