cvg/nice-slam #24: Question on grids initialization

iam4wardto, 2022-07-29:
First I want to say thanks for your work!

It confuses me that during grids initialization, the orders of axis X and Z are all swapped, like the following line does:
https://github.com/cvg/nice-slam/blob/96bb7be8e309a60e2b39515e50ea5cea6efbe449/src/NICE_SLAM.py#L217

Can you help explain why need swapping the axes here, thanks in advance!

Zzh2000, 2022-08-08:
Hi, thanks for your interest in our work!
For this, I actually do not remember carefully and I think maybe the code of this part is not very perfectly written. The high-level goal is to make the axis consistent and let the frustum feature selection and grid sample get the correct features. There is another swap here: https://github.com/cvg/nice-slam/blob/96bb7be8e309a60e2b39515e50ea5cea6efbe449/src/Mapper.py#L317

closed 2022-08-08 by Zzh2000

youmi-zym, 2022-08-09:
Hi @iam4wardto , My understanding is:
The **coarse_val_shape** is generated according to the [bound](https://github.com/cvg/nice-slam/blob/master/src/NICE_SLAM.py#L211), i.e., the bound for the X, Y, and Z axis.
However, the shape of the grid feature should be [B, C, D, H, W]. 
Here, D, H, and W correspond to Z, Y, and X respectively, so we should reverse it like:
coarse_val_shape[0], coarse_val_shape[2] = coarse_val_shape[2], coarse_val_shape[0]