cvg/nice-slam #27: Debug / breakpoint

cazhang, 2022-08-12:
This isn't an issue. 
Just to ask how do you debug (eg. set breakpoint) with this multiprocessing case?

Zzh2000, 2022-08-12:
Hi, 
Maybe you can try to kill the master process if one of the processes encounters an error. Then you will not be bothered by the output of the process that is still running.

closed 2022-08-12 by Zzh2000

cazhang, 2022-08-12:
In the case of run.py, is run.py the master process and mapper and tracker its sub processes? 
does that mean I could kill the main one and then set breakpoint in mapper or tracker?

Zzh2000, 2022-08-16:
You are welcome. Yes, the sub-processes are dispatched here https://github.com/cvg/nice-slam/blob/7af15cc33729aa5a8ca052908d96f495e34ab34c/src/NICE_SLAM.py#L294
I think you can try it.