cvg/nice-slam #29: How do you associate the coarse mapping with mapping and tracking?

calmelo, 2022-08-25:
I read your codes, and I can't find the associations with the coarse mapping and other 2 threads. However, in your paper, the coarse mapping is helpful to Tracking thread. 
Also, in 3.3 Mapping, the loss = coarse geometric loss + fine geometric loss + photometric losses. However, I feel that your codes only consider coarse geometric loss in coarse mapping, and only consider mid&fine geometric loss in mapping, so is the formula (10) correct?

Finally, my biggest confusion is that how do you use the information from coarse mapping in tracking thread and mapping thread. In tracking thread, although you update parameters from mapping and update the feature grid, however, the stage of tracking is "color", and I think that tracking cannot use the coarse mapping information. I will be very appreciate if you can tell me about these questions.

Zzh2000, 2022-08-25:
We did not add this function in the open-sourced code version. We already spent quite a lot of time  (3 month+) reformulating and cleaning the code to make the open-sourced code version better. Furthermore, the frame loss experiment is in the supplementary not in the main paper. Hope you can understand. Thanks!

closed 2022-08-25 by Zzh2000

calmelo, 2022-08-25:
> We did not add this function in the open-sourced code version. We already spent quite a lot of time (3 month+) reformulating and cleaning the code to make the open-sourced code version better. [...]

I know your coarse mapping is important and helpful to tracking, however, I can't really find that how tracking thread uses the coarse information. Can you tell me details in your codes?