cvg/nice-slam #32: environment error

anghaocheng, 2022-09-01:
Hi, thank you for such an excellent work.

When I try to install the environment, it gives an error like ResolvePackageNotFound or Found conflicts!

How can I solve these problems or is there other ways to install the environment?

Zzh2000, 2022-09-01:
Hi,
thanks a lot for your interest in our work!
We ran the released code on many devices with different GPUs across institutions and did not encounter the same issue.  The conda environment config file is sufficient. I wonder if you are using the Ubuntu system and installed the environment using the conda config file.
Good luck!

closed 2022-09-01 by Zzh2000

anghaocheng, 2022-09-02:
Thank you very much for your reply!
I used the window system before and there would be errors. When I switch to the Ubuntu system, there is no problem.