cvg/nice-slam #38: marching_cubes error. Possibly no surface extracted from the level set.

DingYikang, 2022-09-19:
Hi, thanks for your great work!

When test the demo data, I meet the `marching_cubes error`, and there is nothing saved in `output/Demo/mesh`. I'd like to know why I met this problem. Looking forward to your reply.

By the way, when I test the Demo scan, it costs more than 1 hour on single 2080Ti GPU  rather than several minutes as you mentioned in README. Is this runtime normal?

closed 2022-09-20 by DingYikang

slam-nerf, 2022-09-25:
你好，请问您的问题最后是怎样解决的呢？我们也遇到了和你类似的问题：在可视化的时候，我们只能可视化相机的轨迹，但是mesh文件是空的

slam333, 2022-10-17:
Hello, I also encountered the problem of not getting mesh when running on RTX2080ti, how did your problem finally be solved? looking forward to your answer.

Zzh2000, 2022-10-17:
Hi! I think directly running the provided data should not have this error. If you are running on your own custom data, then you might need to debug a bit,  if there is no surface (zero-level set) in the scene representation, then there will be this error.

slam333, 2022-10-17:
Thank you very much for your quickly reply, I was directly running the provided data named Replica, then set the mesh_freq to 50 and set the ckpt_freq to 50, after entering the run command "python -W ignore run.py
configs/Replica/room0.yaml ", when the Room0 dataset was run to 55 frames, I was able to get the ".tar" file in the ckpts folder in the output, but there is no ".ply" file in the mesh folder, do you know what is the error? Thank you

csLinZhang, 2023-03-23:
I solved this problem by adding "from skimage import measure as ski_measure(any name you want)" before "import skimage" in the file "src/uitils/Mesher.py". Of course the corresponding usage "skimage.measure.marching_cubes" has to be changed to "ski_measure.marching_cubes". At least this works for me.