cvg/nice-slam #39: running on dataset that does not contain gt_camera_poses

wangm-buaa, 2022-09-22:
Hi, thanks for publishing this code.
I want to use only color images and depth maps for scene reconstruction, and I have set the 'gt_camera' parameter in the configure file to False. Unfortunately it seems that the code has to read the poses information for some calculations. How do I solve this problem?

Zzh2000, 2022-09-22:
Hi, you can check this part in README. https://github.com/cvg/nice-slam#use-your-own-rgb-d-sequence-from-kinect-azure

closed 2022-09-22 by Zzh2000

Zoey6677, 2024-04-18:
> Hi, thanks for publishing this code. I want to use only color images and depth maps for scene reconstruction, and I have set the 'gt_camera' parameter in the configure file to False. Unfortunately [...]

 Hi~@wangm-buaa 
Would you mind sharing your dataset with me? Thank you very much!

RitaRitaki, 2024-10-29:
> Hi, thanks for publishing this code. I want to use only color images and depth maps for scene reconstruction, and I have set the 'gt_camera' parameter in the configure file to False. Unfortunately [...]

Hey did you find any solution on that? I have the same problem...could you manage to obtain the reconstruction?