cvg/nice-slam #42: Render quality drop when gt_depth is not given

desaixie, 2022-09-28:
Thanks for the great work and releasing the code.

Here I am testing the inference(rendering) quality of Nice-SLAM, i.e. loading pretrained decoders and feature grids from the provided checkpoints and then calling `visualizer.vis` with a camera pose as input. I noticed that `gt_depth` is an optional parameter in the method `render_batch_ray`. I obtained the following two results, one with gt_depth passed to `render_batch_ray` and one without, respectively. 
[image]
[image]

The latter seems to have a non-smooth depth prediction and thus non-smooth color prediction as well. When `gt_depth is None`, a chunk of code in `render_batch_ray` at [`Renderer.py:112-150`](https://github.com/cvg/nice-slam/blob/master/src/utils/Renderer.py#L112) is skipped, which I assume is what might caused this (I don't have a good background in NeRF/rendering, so correct me if I am wrong). Are those results expected? If ground truth depth is required at inference time to get good quality predictions, then what's the point of predicting depth?

MarvinChung, 2022-09-28:
Dear [desaixie](https://github.com/desaixie), we have the same observation as yours. The render result is much worse when there is no gt depth provided. We also wonder why the performance drops so much. For your question, the render function should be used in both training and inference. They use geometric loss that requires predicting the depth mentioned in their paper. Therefore, predicting depth is needed. If you want to infer the NeRF model from viewpoints without ground truth depth, I think the quality will drop, as you mentioned. I think the alternative way to see the map from different viewpoints is to use the mesh. They post-process and use marching cubes to generate mesh from NeRF. We can wait for the authors to answer.

Zzh2000, 2022-09-28:
Hi, 
thanks for your interest in our work.
This phenomenon is expected. I must clarify that the gt_depth is just the observed depth from the depth sensor e.g. Kinect Azure. The use of depth during sampling is discussed in the main paper. This strategy will have benefits for the whole system. I strongly suggest future work to consider using it.

closed 2022-09-28 by Zzh2000