cvg/nice-slam #44: how to use 3070ti to run nice-slam

chesily, 2022-10-01:
I run NICE-SLAM on a short ScanNet sequence with 500 frames, following the readme.md, which notes that It takes a few minutes with ~5G GPU memory. Although I have 8G GPU memory, but it occured the error 

> RuntimeError: CUDA error: out of memory

so how can i adopt the hyperparameters to fit my configuration?
I have seen other issues, which you notes that your classmates have run successfully in 1080ti. I think my configuration is enough, right?

pengsongyou, 2022-10-01:
Have you tried to run with the demo sequence that we provided? Is that fitting in your memory? If yes, what config do you run your ScanNet sequence? Maybe you can see the difference between the two config files?

BTW, 1080ti has ~11G GPU memory and I don't think we have tested on any GPUs with 8G memory.

Best
Songyou

chesily, 2022-10-02:
alright, maybe I need to put this project on hold until I update my hardware

I run the results of Nice-SLAM which you provide successfully, but I can not run Nice-SLAM demo

closed 2022-10-02 by chesily

Zzh2000, 2022-10-02:
Maybe you can try out some GPU lease websites, e.g. AutoDL, Amazon AWS. Then you can choose the hardware setting. 
Good luck!

chesily, 2022-10-02:
> 

thanks, I am a beginner, and I thought that only deep learning model training could run on the server.

This is good news for me.