cvg/nice-slam #48: Error: global /io/opencv/modules/core/src/parallel_impl.cpp (240) WorkerThread 30: Can't spawn new thread

D-Blade, 2022-10-18:
Hi,

congrats on your excellent work and thanks a lot for releasing the code. I just successfully tested the program for all datasets provided **except** the scene of self-captured apartment, which throws the error message below:

(nice-slam) root@ubuntu:~/NeRF_SLAM/nice-slam# python -W ignore run.py configs/Apartment/apartment.yaml
INFO: The output folder is output/Apartment
INFO: The GT, generated and residual depth/color images can be found under output/Apartment/tracking_vis/ and output/Apartment/mapping_vis/
INFO: The mesh can be found under output/Apartment/mesh/
INFO: The checkpoint can be found under output/Apartment/ckpt/
[ERROR:0@27.057] global /io/opencv/modules/core/src/parallel_impl.cpp (240) WorkerThread 29: Can't spawn new thread: res = 11
[ERROR:0@27.057] global /io/opencv/modules/core/src/parallel_impl.cpp (240) WorkerThread 30: Can't spawn new thread: res = 11
cannot allocate memory for thread-local data: ABORT
[ERROR:0@24.091] global /io/opencv/modules/core/src/parallel_impl.cpp (240) WorkerThread 30: Can't spawn new thread: res = 11
[ERROR:0@24.091] global /io/opencv/modules/core/src/parallel_impl.cpp (240) WorkerThread 31: Can't spawn new thread: res = 11
[ERROR:0@24.091] global /io/opencv/modules/core/src/parallel_impl.cpp (240) WorkerThread 32: Can't spawn new thread: res = 11
[ERROR:0@24.091] global /io/opencv/modules/core/src/parallel_impl.cpp (240) WorkerThread 33: Can't spawn new thread: res = 11
[ERROR:0@24.091] global /io/opencv/modules/core/src/parallel_impl.cpp (240) WorkerThread 34: Can't spawn new thread: res = 11
[ERROR:0@24.091] global /io/opencv/modules/core/src/parallel_impl.cpp (240) WorkerThread 35: Can't spawn new thread: res = 11
[ERROR:0@24.092] global /io/opencv/modules/core/src/parallel_impl.cpp (240) WorkerThread 36: Can't spawn new thread: res = 11
[ERROR:0@24.092] global /io/opencv/modules/core/src/parallel_impl.cpp (240) WorkerThread 37: Can't spawn new thread: res = 11
[ERROR:0@24.092] global /io/opencv/modules/core/src/parallel_impl.cpp (240) WorkerThread 38: Can't spawn new thread: res = 11
[ERROR:0@24.092] global /io/opencv/modules/core/src/parallel_impl.cpp (240) WorkerThread 39: Can't spawn new thread: res = 11
[ERROR:0@24.092] global /io/opencv/modules/core/src/parallel_impl.cpp (240) WorkerThread 40: Can't spawn new thread: res = 11
[ERROR:0@24.092] global /io/opencv/modules/core/src/parallel_impl.cpp (240) WorkerThread 41: Can't spawn new thread: res = 11
[ERROR:0@24.092] global /io/opencv/modules/core/src/parallel_impl.cpp (240) WorkerThread 42: Can't spawn new thread: res = 11
[ERROR:0@24.092] global /io/opencv/modules/core/src/parallel_impl.cpp (240) WorkerThread 43: Can't spawn new thread: res = 11
[ERROR:0@24.092] global /io/opencv/modules/core/src/parallel_impl.cpp (240) WorkerThread 44: Can't spawn new thread: res = 11
[ERROR:0@24.092] global /io/opencv/modules/core/src/parallel_impl.cpp (240) WorkerThread 45: Can't spawn new thread: res = 11
[ERROR:0@24.092] global /io/opencv/modules/core/src/parallel_impl.cpp (240) WorkerThread 46: Can't spawn new thread: res = 11
[ERROR:0@24.092] global /io/opencv/modules/core/src/parallel_impl.cpp (240) WorkerThread 47: Can't spawn new thread: res = 11
[ERROR:0@24.092] global /io/opencv/modules/core/src/parallel_impl.cpp (240) WorkerThread 48: Can't spawn new thread: res = 11
[ERROR:0@24.092] global /io/opencv/modules/core/src/parallel_impl.cpp (240) WorkerThread 49: Can't spawn new thread: res = 11
[ERROR:0@24.092] global /io/opencv/modules/core/src/parallel_impl.cpp (240) WorkerThread 50: Can't spawn new thread: res = 11
[ERROR:0@24.092] global /io/opencv/modules/core/src/parallel_impl.cpp (240) WorkerThread 51: Can't spawn new thread: res = 11
[ERROR:0@24.092] global /io/opencv/modules/core/src/parallel_impl.cpp (240) WorkerThread 52: Can't spawn new thread: res = 11
[ERROR:0@24.092] global /io/opencv/modules/core/src/parallel_impl.cpp (240) WorkerThread 53: Can't spawn new thread: res = 11
[ERROR:0@24.092] global /io/opencv/modules/core/src/parallel_impl.cpp (240) WorkerThread 54: Can't spawn new thread: res = 11
[ERROR:0@24.092] global /io/opencv/modules/core/src/parallel_impl.cpp (240) WorkerThread 55: Can't spawn new thread: res = 11
[ERROR:0@24.092] global /io/opencv/modules/core/src/parallel_impl.cpp (240) WorkerThread 56: Can't spawn new thread: res = 11
[ERROR:0@24.092] global /io/opencv/modules/core/src/parallel_impl.cpp (240) WorkerThread 57: Can't spawn new thread: res = 11
[ERROR:0@24.092] global /io/opencv/modules/core/src/parallel_impl.cpp (240) WorkerThread 58: Can't spawn new thread: res = 11
[ERROR:0@24.092] global /io/opencv/modules/core/src/parallel_impl.cpp (240) WorkerThread 59: Can't spawn new thread: res = 11
terminate called after throwing an instance of 'std::bad_alloc'
[ERROR:0@24.092] global /io/opencv/modules/core/src/parallel_impl.cpp (240) WorkerThread 60: Can't spawn new thread: res = 11
terminate called recursively
  what():  std::bad_alloc

I'm testing the program on a remote server with Ubuntu 20.04 and a NVIDIA A100 80GB PCIe GPU, so it's certainly not a memory related issue.
Any hints or suggestions would be helpful!

Zzh2000, 2022-10-18:
Hi,
I actually never encounter this problem. I searched on the internet, one possible solution might be adding:
import cv2
cv2.setNumThreads(1)
Let me know if it helps.

closed 2022-10-18 by Zzh2000

D-Blade, 2022-10-20:
Unfortunately it didn't solve the error. It is weird that the error only occurs when running in the Apartment scene. Have you configured anything specific related to worker threads when using Azure data sequences in the code?

D-Blade, 2022-10-20:
Got it. Thanks though for such a quick reply!

tiemuer1998, 2023-08-11:
> Got it. Thanks though for such a quick reply!

Hi, I also encountered the same problem by adding a line of code to change the environment variable based on the author's suggestion: 

import os
os. environment ['OMP_NUM_THREADS']='4'

The program is running fine now
hope it can help you！

Juanita130, 2024-09-20:
import os
os. environ ['OMP_NUM_THREADS']='4'

Juanita130, 2024-09-20:
add in run.py
import os
os. environ ['OMP_NUM_THREADS']='4'