cvg/nice-slam #49: No coarse decorder is used in tracker

anghaocheng, 2022-10-19:
Hi, thank you for such an excellent work.
The paper says due to the coarse feature grid, "This extrapolated geometry provides a meaningful signal for the tracking as the camera moves into previously unobserved areas. Making it more robust to sudden frame loss or fast camera movement."
But, When I read the code, I find in Tracker.py, it just use color stage decorde to compute the loss and update camera pose. No coarse grid and decorder is fined. 
I would appreciate it if you could answer this question

anghaocheng, 2022-10-19:
I got it. It's really hard work. Sincerely thank you for your work and quick reply！

TontonTremblay, 2022-10-27:
@Zzh2000 would you mind elaborating how this change could impact the performance of the algorithm.

Zzh2000, 2022-10-27:
Sure. During frame loss (or sudden camera movement) the coarse level's prediction is able to provide useful information for tracking, shown in the supplementary and the last video of the project page. For normal use cases, it will not have too much of a difference.