cvg/nice-slam #50: mesh has no attribute vertices

XilinGong, 2022-10-23:
when I run the demo, it shows the problem as follows:

  File "nice-slam-master/src/Mapper.py", line 646, in run
    clean_mesh=self.clean_mesh, get_mask_use_all_frames=False)
  File "nice-slam-master/src/utils/Mesher.py", line 509, in get_mesh
    vertices = mesh.vertices
AttributeError: 'list' object has no attribute 'vertices'

Why mesh was treated as a list?

Zzh2000, 2022-10-23:
I am not sure if you modified anything or followed the install guide. The code is tested by a lot of students. It should work fine.

closed 2022-10-23 by Zzh2000

TontonTremblay, 2022-10-27:
This happens when the algorithm was not able to create a mesh from the mapping step. Depending on the data you use and how hard it is to find texture, you might not be able to create a mesh. You can change the mesh_freq in the config file to something greater: 
```
mapping:
  every_frame: 10
  vis_freq: 50
  vis_inside_freq: 30
  mesh_freq: 50
  ```

XilinGong, 2022-11-01:
> 

Thanks! This really helps a lot.

TontonTremblay, 2022-11-01:
if you check the fork we did, we fixed a lot of code instability issues.
https://github.com/TontonTremblay/nice-slam it might help you