cvg/nice-slam #52: Question on loading GT pose

ckLibra, 2022-10-30:
Hi, thanks for your brilliant job! As mentioned in Issue 10, the camera pose[:3, 1:2] should be multiply by -1 as:
c2w[:3, 1] *= -1.0
c2w[:3, 2] *= -1.0

But I am confused if this is because of the alignment of axis, why do you just convert the rotation but not also the translation?

Zzh2000, 2022-10-30:
I am not sure what you mean by translation. Here we change to the rotation axis because of the different coordinate system between the provided dataset and nerf pixel sampling's default coordinate system. Maybe you can consult your CV course teacher for help.

closed 2022-10-30 by Zzh2000

ckLibra, 2022-10-30:
The c2w transformation matrix is 4*4, including [R t]. You only multiply the -1.0 in the rotation part R. My question is why the axis alignment is only done in R, but not also in t, i.e. c2w[1:3, 3] *= -1.