cvg/nice-slam #53: Error for function render_batch_ray, sometimes met empty gt_depth

moyutianque, 2022-10-31:
Thanks for the nice project.
I have met one issue. When I was training with scannet scene0000_00, there is an error in file utils/Renderer.py.
```
line 112, in render_batch_ray
    far = torch.clamp(far_bb, 0,  torch.max(gt_depth*1.2))
RuntimeError: max(): Expected reduction dim to be specified for input.numel() == 0. Specify the reduction dim with the 'dim' argument.
```
I use the demo.yaml and only replace the data src to the ScanNet data which is preprocessed following ScanNet/SenseReader 
The running script is 
````
python -W ignore run.py configs/Demo/demo.yaml --nice
```

Zzh2000, 2022-11-01:
Could you print out the shape and some value of gt_depth?

Zzh2000, 2022-11-01:
I see, I never encountered this in my experiments, did you modify something? gt_depth is the depth sensor output, which should not be empty.

moyutianque, 2022-11-01:
I discarded all my changes, but this issue still exists. inside_mask in Mapper around line 477 sometimes are all False values.

Zzh2000, 2022-11-06:
Sorry, I don't know. I asked others and they also did not encounter this problem. I  would suggest that you don't do the clamping, and directly assign far = far_bb or assign all same values e.g. 5m for far. Then there won't be an error.

closed 2022-11-06 by Zzh2000

szgy66, 2022-12-19:
I have the same problem. Have you solved it？

moyutianque, 2023-01-11:
> 

I have not solved

moyutianque, 2023-01-11:
> I have the same problem. Have you solved it？

I have not solved

szgy66, 2023-01-12:
> > I have the same problem. Have you solved it？
> 
> I have not solved

I had solved. It may be Render loss too large, you should give the groundtruth

lixinhaiyk, 2023-02-26:
> > > I have the same problem. Have you solved it？
> > 
> > 
> > I have not solved
> 
> I had solved. It may be Render loss too large, you should give the groundtruth

bro,i also meet this problem,you means the groundtruth is lost for this picture? thanks a lot

szgy66, 2023-02-27:
> > > > I have the same problem. Have you solved it？
> > > 
> > > 
> > > I have not solved
> > 
> > 
> > I had solved. It may be Render loss too large, you should give the groundtruth
> 
> bro,i also meet this problem,you means the groundtruth is lost for this picture? thanks a lot

Yes, although the code can run only RGB examples, the effect is very poor. When loss is superimposed to a certain extent, the model can no longer be optimized and can only be interrupted. So you need to provide real pose, like TUMD data set, and it works fine.

Lijingbohere, 2023-06-02:
"So you need to provide real pose, like TUMD data set, and it works fine."What does this phrase mean brother, isn't RGB, depth, pose already passed in when reading data

Lijingbohere, 2023-06-02:
[image]
this is my problem