cvg/nice-slam #60: Doubt about intrinsics of the Apartment dataset.

TwiceMao, 2022-11-26:
I notice a mismatch in the Apartment dataset.
The values of H, W, cx, cy are 720, 1280, 636.9967041015625 and 369.2689514160156 corespondingly.
The quotient of H divided by W is 1.777777777777778.
The quotient of  cx divided by cy is 1.725021022371109.
The difference between these two values is about 0.052.
And the config file of the Apartment dataset does not contain crop_size and crop_edge.

To some extent, I think this leads inaccuracy of the mapping from a image to 3-dimension points.

So, do you handle it somewhere else?
Or could you please help me  understand it? Thank you.

Zzh2000, 2022-11-26:
After the first step in https://github.com/cvg/nice-slam#use-your-own-rgb-d-sequence-from-kinect-azure, open3d will automatically generate an intrinsic YAML file. We directly take the parameters from this file. 
I don't know if it is true that the quotient of H divided by W should be the same as the quotient of cx divided by cy, could you show me with some references?

closed 2022-11-26 by Zzh2000

TwiceMao, 2022-11-27:
Sorry, I wanted to say that the quotient of W divided by H should be the same as the quotient of cx divided by cy.

I mistakenly referred to the Apartment dataset as a synthetic dataset.
If that is the case, it should hold true for a synthetic dataset that the center of the camera is at the center of the image in pixel coordinate.
Then,the quotient of W divided by H should be the same as the quotient of cx divided by cy.
But the Apartment is not a synthetic dataset.