cvg/nice-slam #63: Run own RGB-D sequence from ROS

szgy66, 2022-12-16:
The work is so interesting. It does two things: location and reconstruction. I'd like to replicate your code in our own data set. So I captured a series of RGB-D data sets using the ros camera.Then run the recommended tutorial. However, I found that the Re-rendering loss keeps increasing, and when Tracking Frame 691, the following error is reported.

Tracking Frame  689

Re-rendering loss: 61943169.13->32158803.15 camera tensor error: 2.3272->2.3433

Tracking Frame  690

Re-rendering loss: 14259613.90->6811814.82 camera tensor error: 2.4108->2.4126

Coarse Mapping Frame  690

Mapping Frame  690

Tracking: update the parameters from mapping

Tracking Frame  691

[W CudaIPCTypes.cpp:15] Producer process has been terminated before all shared CUDA tensors released. See Note [Sharing CUDA tensors]
Process Process-1:
Traceback (most recent call last):
  File "/opt/conda/envs/nice-slam/lib/python3.7/multiprocessing/process.py", line 297, in _bootstrap
    self.run()
  File "/opt/conda/envs/nice-slam/lib/python3.7/multiprocessing/process.py", line 99, in run
    self._target(*self._args, **self._kwargs)
  File "/workspace/nice-slam/src/NICE_SLAM.py", line 266, in tracking
    self.tracker.run()
  File "/workspace/nice-slam/src/Tracker.py", line 233, in run
    camera_tensor, gt_color, gt_depth, self.tracking_pixels, optimizer_camera)
  File "/workspace/nice-slam/src/Tracker.py", line 107, in optimize_cam_in_batch
    self.c, self.decoders, batch_rays_d, batch_rays_o,  self.device, stage='color',  gt_depth=batch_gt_depth)
  File "/workspace/nice-slam/src/utils/Renderer.py", line 109, in render_batch_ray
    far = torch.clamp(far_bb, 0,  torch.max(gt_depth*1.2))
RuntimeError: max(): Expected reduction dim to be specified for input.numel() == 0. Specify the reduction dim with the 'dim' argument.

I don't understand why there is such a mistake, can you help to solve it.Thanks!

Zzh2000, 2022-12-20:
Hi, 
Thanks for your interest in our work!
I think you can ignore this specific RuntimeError since it is caused by large camera tracking error. You can focus on making tracking not drifting away. One way to debug is to directly provide pseudo ground truth camera poses (e.g. calculated by Open3D redwood).
Best wishes!

closed 2022-12-20 by Zzh2000

AlvinYH, 2022-12-22:
Hi, thanks for your interesting work! I've encountered the same problem caused by large camera tracking error. I'm wondering if the model can generate satisfying resulting without annotations of pseudo ground truth camera poses, as we'd like to test the algorithm on a pre-recorded video. Also, is there any way to obtain pseudo GT camera poses from the video not recorded by Azure Kinect?

Zzh2000, 2022-12-26:
Hi,
Thanks for your interest in our work!
The pseudo ground truth camera poses will not affect the system's result, but would be great for debugging. 
The capture device does not matter, as long as you can get the depth/color/intrinsics needed for the Redwood in Open3D. https://github.com/cvg/nice-slam#use-your-own-rgb-d-sequence-from-kinect-azure
Best wishes!

Zoey6677, 2024-04-18:
> Hi, thanks for your interesting work! I've encountered the same problem caused by large camera tracking error. I'm wondering if the model can generate satisfying resulting without annotations of [...]

@AlvinYH 
Hi~Do you know how to use the author's tutorial (https://github.com/cvg/nice-slam#use-your-own-rgb-d-sequence-from-kinect-azure) with a different camera? I've used my own camera (Kinect2) to get depth/color/intrinsics, but I'm confused about how to do it next.

Zoey6677, 2024-04-18:
> The work is so interesting. It does two things: location and reconstruction. I'd like to replicate your code in our own data set. So I captured a series of RGB-D data sets using the ros camera.Then [...]

@szgy66  Hi~
Would you mind sharing your dataset with me? Thank you very much!