cvg/nice-slam #64: Run own RGBD dataset with pose

szgy66, 2022-12-27:
Hello!Thank you very much for your project.I tried to replicate your results on my own data set.First, I reproduced the TUMD data according to your README.md instructions, and indeed reconstructed better results.But when I used ORB_SLAM3 to calculate and save the pose information for TUMD and then retrain/visualize, the result was not satisfactory. I use "evo" to compare two tracks as shown below, while I printed the value of c2w in dataset.py(left is the officially provided groundtruth.txt, right is my own). The third image is the ORB computed saved pose.

I really don't understand why the pose is the same height, but the reconstruction results are quite different.
Can you help me? Thanks!
 
[image].

[image]

[image]

Zzh2000, 2022-12-29:
Hi, 
If the poses are nearly the same, there should not be too much reconstruction difference.

closed 2022-12-29 by Zzh2000

anghaocheng, 2023-03-09:
> Hello!Thank you very much for your project.I tried to replicate your results on my own data set.First, I reproduced the TUMD data according to your README.md instructions, and indeed reconstructed [...]

Hello, I also encountered this problem when using orb-slam2 to do similar operations. How do you solve it? Is it related to the direction defined by the camera coordinate system in pytorch, which is different from orb-slam

anghaocheng, 2023-03-09:
Hello, I also encountered this problem when using orb-slam2 to do similar operations. How do you solve it? Is it related to the direction defined by the camera coordinate system in pytorch, which is different from orb-slam