cvg/nice-slam #72: Pytorch RuntimeError

Sangmin-Bak, 2023-02-22:
Thank you for your research, I have a question because the following issue arose.
When I run "run.py", I get a RuntimeError like below. I would like to know how to solve this.

Traceback (most recent call last):
  File "/home/psm/anaconda3/envs/nice-slam/lib/python3.7/multiprocessing/process.py", line 297, in _bootstrap
    self.run()
  File "/home/psm/anaconda3/envs/nice-slam/lib/python3.7/multiprocessing/process.py", line 99, in run
    self._target(*self._args, **self._kwargs)
  File "/home/psm/workspace/nice-slam/src/NICE_SLAM.py", line 276, in mapping
    self.mapper.run()
  File "/home/psm/workspace/nice-slam/src/Mapper.py", line 606, in run
    gt_c2w, self.keyframe_dict, self.keyframe_list, cur_c2w=cur_c2w)
  File "/home/psm/workspace/nice-slam/src/Mapper.py", line 518, in optimize_map
    val[mask] = val_grad.clone().detach()
RuntimeError: out_ptr == out_accessor[thread_count_nonzero[tid + 1]].data()INTERNAL ASSERT FAILED at "/opt/conda/conda-bld/pytorch_1634272168290/work/aten/src/ATen/native/TensorAdvancedIndexing.cpp":1723, please report a bug to PyTorch.

Zzh2000, 2023-02-22:
Hi, please make sure you are using the exact conda environment.yaml and ubuntu desktop. If all versions match, there should not be a problem.

closed 2023-02-22 by Zzh2000

Zoey6677, 2024-01-23:
@Sangmin-Bak Hi,did you meet the following error?
(nice-slam) yang@yang-C9X299-PGF:~/data/zy/nice-slam-master$ python -W ignore run.py configs/Demo/demo.yaml
Traceback (most recent call last):
  File "run.py", line 47, in <module>
    main()
  File "run.py", line 41, in main
    slam = NICE_SLAM(cfg, args)
  File "/home/yang/data/zy/nice-slam-master/src/NICE_SLAM.py", line 75, in __init__
    self.estimate_c2w_list.share_memory_()
  File "/mnt/data/conda_envs/nice-slam/lib/python3.7/site-packages/torch/_tensor.py", line 482, in share_memory_
    self.storage().share_memory_()
  File "/mnt/data/conda_envs/nice-slam/lib/python3.7/site-packages/torch/storage.py", line 480, in share_memory_
    self._storage.share_memory_()
  File "/mnt/data/conda_envs/nice-slam/lib/python3.7/site-packages/torch/storage.py", line 160, in share_memory_
    self._share_filename_()
RuntimeError: torch_shm_manager at "/mnt/data/conda_envs/nice-slam/lib/python3.7/site-packages/torch/bin/torch_shm_manager": could not generate a random directory for manager socket

> Thank you for your research, I have a question because the following issue arose. When I run "run.py", I get a RuntimeError like below. I would like to know how to solve this.
>
> Traceback (most [...]