cvg/nice-slam #74: [Open3D WARNING] Failed to initialize GLFW

lixinhaiyk, 2023-03-03:
hi:
I really appreciate your work,and when I evaluate the recon error with -2d ，I meet the following error:

[Open3D WARNING] GLFW Error: X11: The DISPLAY environment variable is missing
[Open3D WARNING] Failed to initialize GLFW
  File "/root/nice-slam-optimized/src/tools/eval_recon.py", line 157, in calc_2d_metric
    vis.get_render_option().mesh_show_back_face = True
AttributeError: 'NoneType' object has no attribute 'mesh_show_back_face'

I use it on the remote server ,and wonder whether this evaluation need a desktop display?
thanks a lot!

lixinhaiyk, 2023-03-03:
and....can the results in paper about Acc. Comp. Comp.Ratio only use -3d to evaluate ?

Zzh2000, 2023-03-03:
Hi!
Yes, you need a desktop display for 2d evaluation since it uses open3d rendering, and open3d rendering in default does not work on server. For the second question, yes, -3d is OK.

closed 2023-03-03 by Zzh2000

Zoey6677, 2024-03-24:
> hi: I really appreciate your work,and when I evaluate the recon error with -2d ，I meet the following error:
>
> [Open3D WARNING] GLFW Error: X11: The DISPLAY environment variable is missing [Open3D [...]

Hi~@lixinhaiyk
How did you solve it in the end?