cvg/nice-slam #79: Question about code to pre-filter out of bounding box values

TomCC7, 2023-03-25:
https://github.com/cvg/nice-slam/blob/7af15cc33729aa5a8ca052908d96f495e34ab34c/src/Tracker.py#L94-L104

I'm having some trouble trying to understand the method to create masks to make sure ray ends within the mapping bound. We need the end-points of rays to satisfy that
```
b_l <= depth \cdot dir + origin <= b_u
```

I'm confused why the min-max operation here is correct. Is this a bug?
https://github.com/cvg/nice-slam/blob/7af15cc33729aa5a8ca052908d96f495e34ab34c/src/Tracker.py#L99

My implementation is provided below:
```python
            with torch.no_grad():
                # pre-filter those out of bounding box depth value for niceslam
                # b_l <= depth \cdot dir + origin <= b_u
                det_rays_o = batch_rays_o.clone().detach().unsqueeze(-1)  # (N, 3, 1)
                det_rays_d = batch_rays_d.clone().detach()
                # normalize ray direction
                det_rays_d /= torch.linalg.vector_norm(det_rays_d, dim=0)
                # bounds in ray coordinate
                bounds = self.bound.unsqueeze(0).to(device) - det_rays_o
                end_points = batch_gt_depth.unsqueeze(-1) * det_rays_d
                # both sides should be inside the bounding box
                left_mask = (end_points >= bounds[..., 0]).all(dim=1)
                right_mask = (end_points <= bounds[..., 1]).all(dim=1)
                inside_mask = (left_mask & right_mask).squeeze()
```

Zzh2000, 2023-03-25:
I double checked, I think it is correct.

closed 2023-03-25 by Zzh2000

TomCC7, 2023-03-25:
But the generated ray direction isn't even normalized?