cvg/nice-slam #85: How to use D435i to build my own dataset using the method you mentioned

Jameskobedurrant, 2023-04-28:
Hi, thank you very much for your great work first.
I used D435i to build a TUM_ Similar to RGBD dataset, GT uses colmap output pose and modifies freiburg1_desk.yaml successfully ran my own dataset, but the results were very poor. What is the reason for this? How to use D435i to build my own dataset using the method you mentioned?

Zzh2000, 2023-04-28:
Hi,
one thing to debug is to provide the gt pose to NICE_SLAM and only run the mapping, to see if the bad reconstruction result is from the tracking or the mapping. 
Generally, D435i should work, maybe you can also double-check the camera axis convention, gt depth scale (png to float), depth color consistency, intrinsics calibration, and other hyperparameters.

Zoey6677, 2024-04-06:
Hi~@Jameskobedurrant 
Would you mind sharing your dataset with me? As a beginner, I appreciate it very much!

Zoey6677, 2024-04-17:
> Hi, thank you very much for your great work first. I used D435i to build a TUM_ Similar to RGBD dataset, GT uses colmap output pose and modifies freiburg1_desk.yaml successfully ran my own dataset, [...]

Hi~ @Jameskobedurrant 
Would you mind sharing your dataset with me? As a beginner, I appreciate it very much!

Afterg10w, 2024-04-29:
> Zoey6677

Hi,  I had the same problem as you. Did you solve it? I found that when I change the parameter png_depth_scale to 5000 the algorithm works but very poorly, when I set it to 1000 the algorithm reports an error at frame 55, is your situation the same as mine?

Afterg10w, 2024-04-29:
> Hi, thank you very much for your great work first. I used D435i to build a TUM_ Similar to RGBD dataset, GT uses colmap output pose and modifies freiburg1_desk.yaml successfully ran my own dataset, [...]

Hi, I had the same problem as you. Did you solve it? I found that when I change the parameter png_depth_scale to 5000 the algorithm works but very poorly, when I set it to 1000 the algorithm reports an error at frame 55, is your situation the same as mine?