cvg/nice-slam #86: About reproducing nice-slam

Lijingbohere, 2023-05-05:
(nice-slam) lijingbo@Server30901:~/nice-slam-master$ python visualizer.py configs/ScanNet/scene0000.yaml --input  /data_home/home/lijingbo/scene0000_00.zip --output output/vis/scannet/scans/scene0000_00
Traceback (most recent call last):
  File "visualizer.py", line 55, in <module>
    estimate_c2w_list[:, :3, 3] /= scale
NameError: name 'estimate_c2w_list' is not defined

Zzh2000, 2023-05-05:
Please follow the command in README, passing zip file as input is strange.

closed 2023-05-05 by Zzh2000

random-guest, 2023-07-03:
@Zzh2000  I followed the command in README,  but got the following error:

python visualizer.py configs/Replica/room0_imap.yaml --imap 
Traceback (most recent call last):
  File "visualizer.py", line 55, in <module>
    estimate_c2w_list[:, :3, 3] /= scale
NameError: name 'estimate_c2w_list' is not defined

it seems like visualizer.py expects ckpts to be passed as well ?

but when I ran 
python -W ignore run.py configs/Replica/room0_imap.yaml --imap 

it worked properly. 

any suggestions?
thanks in advance.

Zoey6677, 2024-01-22:
> @Zzh2000 I followed the command in README, but got the following error:
> 
> python visualizer.py configs/Replica/room0_imap.yaml --imap Traceback (most recent call last): File "visualizer.py", line 55, in estimate_c2w_list[:, :3, 3] /= scale NameError: name 'estimate_c2w_list' is not defined
> 
> it seems like visualizer.py expects ckpts to be passed as well ?
> 
> but when I ran python -W ignore run.py configs/Replica/room0_imap.yaml --imap
> 
> it worked properly.
> 
> any suggestions? thanks in advance.
Hi,did you meet the the following error:
python visualizer.py configs/Demo/demo.yaml
Traceback (most recent call last):
  File "visualizer.py", line 36, in <module>
    cfg = config.load_config(
AttributeError: module 'src.conv_onet.config' has no attribute 'load_config'

and I ran
python -W ignore run.py configs/Demo/demo.yaml
the same error

thanks in advance

random-guest, 2024-01-22:
hi @Zoey6677 
No, I didn't.