cvg/nice-slam #103: Demo       Stuck at 16 frames, GPU not working

glaciomarine, 2023-10-10:
INFO: The output folder is output/Demo
INFO: The GT, generated and residual depth/color images can be found under output/Demo/vis/
INFO: The mesh can be found under output/Demo/mesh/
INFO: The checkpoint can be found under output/Demo/ckpt/
Tracking Frame 16:   3%|██▎   

Stuck at 16 frames, GPU not working

Zzh2000, 2023-10-17:
Hi, please double-check your environment.

closed 2023-10-17 by Zzh2000

tyz1030, 2023-10-17:
encounting same issue, demo stuck at frame 16. Conda environment built with no issue

tyz1030, 2023-10-19:
actually this bug is a bit weired. It stuck at certain frames on my desktop, but works just fine on my laptop. which is annoying. Both using ubuntu 22.04 with nvidia 4090 series graphics. Not sure what makes the difference

glaciomarine, 2023-10-26:
Thanks for your answer. Now I have encountered a new problem. I will encounter two problems when using 2080ti or 3090, one is CUDA out of memory. The other is thread_monitor Resource temporarily unavailable in pthread_create

 
Closed #103 as completed.

doginono, 2023-11-15:
@glaciomarine How were you able to solve this problem? While trying out the Demo like you, I also got stuck at the 16th frame.