cvg/nice-slam #106: Whether the repository contains training code.

QQQ300kuai, 2023-11-02:
Sorry, I'd like to know if the training code is included in the repository. I want to start training the model again.

Zzh2000, 2023-11-28:
Sorry, the training code is not included, but it should not be that hard, you can directly replace the MLP class in ConvOccNet with NICE-SLAM's MLP and begin training using the default hyperparameter.

closed 2023-11-28 by Zzh2000

Zoey6677, 2024-03-20:
> Sorry, I'd like to know if the training code is included in the repository. I want to start training the model again.

@QQQ300kuai Hello, I would also like to try training the model. I'm a beginner, so it's too difficult for me to understand the author's answer. Can you share how you do that? It would be nice to be more detailed. Thank you so much!